google-coral/pycoral #98: AttributeError: 'Interpreter' object has no attribute '_get_full_signature_list'

lavarok, 2022-12-10:
### Description

I can't get the detect_image.py example to run. I've tried using different models, but I keep getting an error about the signature list. 

### Issue Type

Bug, Build/Install

### Operating System

Ubuntu

### Coral Device

M.2 Accelerator A+E

### Programming Language

Python 3.8

### Relevant Log Output

```shell
----INFERENCE TIME----
Note: The first inference is slow because it includes loading the model into Edge TPU memory.
Traceback (most recent call last):
  File "examples/detect_image.py", line 108, in <module>
    main()
  File "examples/detect_image.py", line 87, in main
    objs = detect.get_objects(interpreter, args.threshold, scale)
  File "/usr/lib/python3/dist-packages/pycoral/adapters/detect.py", line 204, in get_objects
    signature_list = interpreter._get_full_signature_list()
AttributeError: 'Interpreter' object has no attribute '_get_full_signature_list'
```

hjonnala, 2022-12-12:
> ```shell
> AttributeError: 'Interpreter' object has no attribute '_get_full_signature_list'
> ```

Please check the tflite runtime version and if its not `2.5.0post1`, please uninstall and install it again with the instuctions at: https://coral.ai/software/#debian-packages

```
hemanthj@:~$ python3 -c "import tflite_runtime as tflite; print('tflite runtime vesrion:', tflite.__version__);import pycoral; print('pycoral version:', pycoral.__version__)"
tflite runtime vesrion: 2.5.0.post1
pycoral version: 2.0.0
```

lavarok, 2022-12-12:
Thanks for that! While not technically an install issue, it led me to find that I had a copy of tflite_runtime hiding in /usr/local/lib/python3.8.

closed 2022-12-12 by lavarok