google-coral/pycoral #104: Failed precondition: Package requires runtime version (14), which is newer than this runtime version (13).

puelon, 2023-01-19:
### Description

I have a Google Coral USB device and I am currently trying to use it on my Windows 11 machine for a custom object detection model. The model for object detection was trained following this documentation: https://colab.research.google.com/github/khanhlvg/tflite_raspberry_pi/blob/main/object_detection/Train_custom_model_tutorial.ipynb.  

During the installation guide in the Coral.Ai website, I had failed attempts to install  edgetpu_runtime_20221024 version, as it starts but when clicking executed the screen disappears, I searched a bit and installed the edgetpu_runtime_20210119-1 successfully. Therefore, by the end of the tutorial I was able to run both image classification and object detection examples. However, when I use my own model trained on the tutorial above, with the following command:
```

py -3.9 examples/detect_image.py \
  --model test_data/custom_model_edgetpu.tflite \
  --labels test_data/Labels.txt \
  --input test_data/test_image.jpeg \
  --output test_data/test_image.jpeg
```
I am getting a runtime version installed, despite being able to run inference on the downloaded examples. 

### Issue Type

Build/Install, Support

### Operating System

Windows 10

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

### Relevant Log Output

```shell
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_22332\2010350775.py in <module>
    241 
    242 if __name__ == '__main__':
--> 243   main()

~\AppData\Local\Temp\ipykernel_22332\2010350775.py in main()
    206   interpreter = make_interpreter(model_path, edgetpu)
    207 
--> 208   interpreter.allocate_tensors()
    209 
    210   labels = load_labels(label_path)

~\AppData\Roaming\Python\Python39\site-packages\tflite_runtime\interpreter.py in allocate_tensors(self)
    406   def allocate_tensors(self):
    407     self._ensure_safe()
--> 408     return self._interpreter.AllocateTensors()
    409 
    410   def _safe_to_run(self):
```

hjonnala, 2023-01-19:
Please install latest runtime version and replace libusb with the older version: https://github.com/google-coral/tflite/issues/60#issuecomment-1341113519

puelon, 2023-01-20:
Hi thanks for replying. I am using Windows 11 and the .bat file simply closes after I try to run it. However, the same doesn't happen to older versions

hjonnala, 2023-01-20:
I am able to run .bat file normally without any issues on windows 11. Please try to install using `powershell` as `admin`..

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows

PS C:\Windows\system32> cd ../..
PS C:\> cd users
PS C:\users> cd .\heman\Downloads\edgetpu_runtime_20221024\edgetpu_runtime\
PS C:\users\heman\Downloads\edgetpu_runtime_20221024\edgetpu_runtime> .\install.bat
Warning: During normal operation, the Edge TPU Accelerator may heat up,
depending on the computation workloads and operating frequency. Touching the
metal part of the device after it has been operating for an extended period of
time may lead to discomfort and/or skin burns. As such, when running at the
default operating frequency, the device is intended to safely operate at an
ambient temperature of 35C or less. Or when running at the maximum operating
frequency, it should be operated at an ambient temperature of 25C or less.

Google does not accept any responsibility for any loss or damage if the device
is operated outside of the recommended ambient temperature range.
................................................................................
Would you like to enable the maximum operating frequency for the USB Accelerator? Y/N Y
Installing UsbDk
Installing Windows drivers
Microsoft PnP Utility

Adding driver package:  coral.inf
Driver package added successfully. (Already exists in the system)
Published Name:         oem50.inf

Adding driver package:  Coral_USB_Accelerator.inf
Driver package added successfully. (Already exists in the system)
Published Name:         oem1.inf
Driver package installed on device: USB\VID_18D1&PID_9302\5&335d3493&0&3
Driver package installed on device: USB\VID_18D1&PID_9302\5&335d3493&0&13
Driver package installed on device: USB\VID_18D1&PID_9302\5&335d3493&0&14

Adding driver package:  Coral_USB_Accelerator_(DFU).inf
Driver package added successfully. (Already exists in the system)
Published Name:         oem53.inf
Driver package installed on device: USB\VID_1A6E&PID_089A\5&335d3493&0&13
Driver package installed on device: USB\VID_1A6E&PID_089A\5&335d3493&0&14
Driver package installed on device: USB\VID_1A6E&PID_089A\5&335d3493&0&1
Driver package installed on device: USB\VID_1A6E&PID_089A\5&335d3493&0&3

Total driver packages:  3
Added driver packages:  3
Installing performance counters

Info: Provider {aaa5bf9e-c44b-4177-af65-d3a06ba45fe7} defined in C:\users\heman\Downloads\edgetpu_runtime_20221024\edgetpu_runtime\third_party\coral_accelerator_windows\coral.man is already installed in system repository.
Info: Successfully installed performance counters in C:\users\heman\Downloads\edgetpu_runtime_20221024\edgetpu_runtime\third_party\coral_accelerator_windows\coral.manCopying edgetpu and libusb to System32
        1 file(s) copied.
        1 file(s) copied.
Install complete
PS C:\users\heman\Downloads\edgetpu_runtime_20221024\edgetpu_runtime>
```

puelon, 2023-01-21:
it works, thanks a lot!

closed 2023-01-21 by puelon