google-coral/pycoral #105: Does this work on micropython?

achapkowski, 2023-02-09:
### Description

On the microcontroller offering, does it work with micropython?
I couldn’t find any documentation on this.

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

Dev Board

### Other Devices

Rapsberry Pi 4

### Programming Language

Other

hjonnala, 2023-02-09:
> On the microcontroller offering, does it work with micropython?

Hi, which microcontroller are you referring to here?

achapkowski, 2023-02-09:
I'm referring to this board: https://coral.ai/products/dev-board-micro/

hjonnala, 2023-02-09:
No, it doesn't work with micropython. It works with CoralMicro API: https://coral.ai/docs/dev-board-micro/get-started/#the-software

achapkowski, 2023-02-10:
@hjonnala thank you, are there are any thoughts of adding support for micropython?  I didn't know if there was a roadmap anywhere.

hjonnala, 2023-02-10:
No, we don't have any plans of adding support for micropython.

achapkowski, 2023-02-12:
Ok thank you.  Hopefully you can add it to the roadmap in the future.

closed 2023-02-12 by achapkowski