google-coral/pycoral #106: Help understanding how the edgetpu compiler works

APatel2001, 2023-02-11:
Hello,

My friend and I at Purdue University are working on a project where we want to improve and add functionality to the Google Coral device and we were wondering if someone on this github would be willing to explain to us how the edgetpu_compiler works and we are also wondering if the source code for the compiler is availible anywhere!

Thanks,
Avi

hjonnala, 2023-02-13:
Hello @APatel2001 source code for compiler is not open sourced. Please check the documentation for compiler usage at: https://coral.ai/docs/edgetpu/compiler/. Thanks!

APatel2001, 2023-02-14:
Hello Hemanth, from what I am understanding right now, coral only supports a subset of the tflite operations. I was wondering if it is possible to make more of the tflite operations supported by coral without access to the compiler source code!

hjonnala, 2023-02-14:
> I was wondering if it is possible to make more of the tflite operations supported by coral without access to the compiler source code!

No, its not possible.

jinensetpal, 2023-02-18:
Hey, @hjonnala! Thank you for your prompt responses! Another question we had:

We attempted compiling `libedgetpu` with debug flag. While this worked, and it was able to read the symbols correctly, it caused a segmentation fault on inference. Is this a known issue, and is there any way we can get around this?

hjonnala, 2023-02-21:
> We attempted compiling `libedgetpu` with debug flag. While this worked, and it was able to read the symbols correctly, it caused a segmentation fault on inference. Is this a known issue, and is there any way we can get around this?

Hi, The question is not relevant for this thread. Is it happening for all the models or any particular model? Mostly, it would be an issue with the model. Feel free to test the infrence using the script at: https://github.com/hjonnala/snippets/blob/main/coral_inference.py.  and open a new issue for any questions on this one. Thanks!!

closed 2023-02-21 by hjonnala