google-coral/pycoral #107: ModuleNotFoundError: No module named 'pycoral'

bme323, 2023-03-06:
### Description

Hi all,

I am trying to follow the Coral 'Get started with the USB Accelerator' demo and run the parrot demo (step 3 of part 3), however when executing, I get the error message shown.

[image]

Info:
- Raspberry Pi 4B
- 32-bit arm7l  
- downgraded python 3.7.12 from python 3.9.x

I have tried running the 32-bit whl. file and also changing directory name and continue getting error messages.

Any help appreciated.

Thanks

### Issue Type

Build/Install, Support

### Operating System

Linux

### Other Devices

Rapsberry Pi 4

### Programming Language

Python 3.7, Python 3.9

### Relevant Log Output

```shell
Traceback (most recent call last):
  File "examples/classify_image.py", line 37, in <module>
    from pycoral.adapters import classify 
ModuleNotFoundError: No module named 'pycoral'
```

hjonnala, 2023-03-07:
Hi, can you please share the output of below command as root user and without root user..

```
hemanthj@:~$ python3 -c "import tflite_runtime as tflite; print('tflite runtime vesrion:', tflite.__version__);import pycoral; print('pycoral version:', pycoral.__version__)"
tflite runtime vesrion: 2.5.0.post1
pycoral version: 2.0.0
```

bme323, 2023-03-08:
Hi Hemanth,

Here is my output as root user and non-root user. I have not installed tflite. Is this a prerequisite for the https://coral.ai/docs/accelerator/get-started/#pycoral-on-linux demo? 

[image]
[image]

hjonnala, 2023-03-08:
Could you please uninstall the exsiting packages and try the installation again. Please share the installation logs and output of below command. Thanks!!

`python3 -c "import tflite_runtime as tflite; print('tflite runtime vesrion:', tflite.__version__);import pycoral; print('pycoral version:', pycoral.__version__)"`

closed 2023-03-11 by bme323

bme323, 2023-03-11:
Hi thanks,

I reset rpi OS. think initial problem was the pyenv directory but have now run into a new issue about failing to load delegate from libedgetpu. I've created a new issue, and have closed this. Thanks