google-coral/pycoral #108: ValueError: Failed to load delegate from libedgetpu.so.1

bme323, 2023-03-11:
### Description

Hi,

Attempting to follow the Get Started with Coral USB accelerator https://coral.ai/docs/accelerator/get-started/#3-run-a-model-on-the-edge-tpu and getting the error output 'ValueError: Failed to load delegate from libedgetpu.so.1' as shown.

[image]

I have attempted to follow issues with the same error but I get error messages when trying to add the apex user using the command 'sudo adduser root apex'.

I am running on Raspberry Pi 4B, arm7l 32-bit, Python 3.9.2

Thank you

### Issue Type

Bug, Build/Install, Support

### Operating System

Linux

### Coral Device

USB Accelerator

### Other Devices

Rapsberry Pi 4

### Programming Language

Python 3.9

hjonnala, 2023-03-13:
Could you please share the output of below commands:

`python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'`

```
sudo python3 examples/classify_image.py \
--model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--labels test_data/inat_bird_labels.txt \
--input test_data/parrot.jpg
```

bme323, 2023-03-13:
[image]

[image]

bme323, 2023-03-13:
I get the same initial error message when I run the edge tpu parrot example after running the commands you suggested.

CristiFati, 2023-04-07:
Make sure the device is connected and visible.
For some reason on my laptop, if I connect it to a certain *USB* port I get this error, but if I move it to another, it works fine.

hjonnala, 2023-04-27:
Feel free to reopen if issue still exists. Thanks!!

closed 2023-04-27 by hjonnala