google-coral/pycoral #113: Unresponsive Coral Dev Board when running segmentation models

Jaquetti, 2023-05-24:
### Description

Hello,

I am encountering an issue when attempting to run segmentation models from the Coral AI website (https://coral.ai/models/semantic-segmentation/) on my Coral Dev Board. Specifically, when I execute the Python script with the segmentation model, the Coral Dev Board becomes unresponsive, requiring a restart. It's important to note that I do not experience this issue when running detection or classification models.

Python Library Versions:

    Python: 3.7.3
    tflite-runtime: 2.5.0.post1
    pycoral: 2.0.0
    libedgetpu1-std: 16.0
    

### Issue Type

Others

### Operating System

Mendel Linux

### Coral Device

Dev Board

### Programming Language

Python 3.7

hjonnala, 2023-05-24:
can you please share the python script and model you are trying to execute or share the reproducable steps here. Thanks!

Jaquetti, 2023-05-25:
[test_py.zip](https://github.com/google-coral/pycoral/files/11564937/test_py.zip)

This zip file includes my Python script, the weights, and a text file containing information about the models.

Thank you.

hjonnala, 2023-05-25:
The script is working fine on my Coral Dev Board. Please try updating the [board](https://coral.ai/docs/dev-board/reflash/#update-your-board-with-apt-get). Thanks!!

mendel@mocha-llama:~/test_py$ python3 main.py --model mobilenet_segnet_quant_edgetpu.tflite 
Invoke OK!!
mendel@mocha-llama:~/test_py$

Jaquetti, 2023-05-29:
Hello,

I have updated the mendel version 5.2 to 5.3 but the problem still occurs. I'm adressing a video of the tests:

https://drive.google.com/drive/u/1/folders/1rGyOrEEfOsjg4-g4bS875IpSjkTru4ZD

Thank you.

Jaquetti, 2023-06-15:
The problem was my power source.

closed 2023-06-15 by Jaquetti