google-coral/pycoral #114: Segmentation Models label size mismatch in output tensors

stakach, 2023-05-26:
### Description

Downloading the segmentation models from the [coral.ai website](https://coral.ai/models/all/#segmentation) they seem to have an output tensor that matches the input tensor.
i.e. input: 1x256x256x3, output: 1x256x256x3

However all the examples expect there to be an output per-label per-pixel.
i.e. input: 1x256x256x3, output: 1x256x256x21

and looking at tutorials on the web from a few years ago, they all seem to match the example code and the models they are using all work.

So my question is, how does one interpret the output of the new segmentation models?

### Issue Type

Documentation Bug

hjonnala, 2023-05-30:
> matches the input tensor. i.e. input: 1x256x256x3, output: 1x256x256x3
> 
> However all the examples expect there to be an output per-label per-pixel. i.e. input: 1x256x256x3, output: 1x256x256x21

Coral models are traind to classify 3 classes (1x256x256x3) and other models might have trained to classify 21 classes(1x256x256x21).

There can be argamax applied on the final layer for some models. If not [argmax](https://github.com/google-coral/pycoral/blob/master/examples/semantic_segmentation.py#L120) would be applied in post processing.
[image]

stakach, 2023-05-30:
I can use the argmax models without issue, however the labels file on the website includes many more than 3 classes
https://raw.githubusercontent.com/google-coral/test_data/master/pet_labels.txt