google-coral/pycoral #117: Windows 11 AMD64 cannot install with errors.

misosoup7, 2023-06-28:
### Description

Hi I'm running a Intel Core i7 13700KF on Windows 11, Python 3.11.

I'm trying to install PyCoral per the instructions here:
https://coral.ai/docs/m2/get-started/#pycoral-on-win

The issue I'm running into is that running this command gets an error,.
`python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0`
```ERROR: Could not find a version that satisfies the requirement pycoral~=2.0 (from versions: 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9, 0.1.0)
ERROR: No matching distribution found for pycoral~=2.0```

Looking online I see someone with a M1 Mac had this issue last year and one of the recommendation was to install form the wheel files so I downloaded the latest version:
`pycoral-2.0.0-cp39-cp39-win_amd64.whl`

But running
` python3 -m pip install pycoral-2.0.0-cp39-cp39-win_amd64.whl`
results in:
`ERROR: pycoral-2.0.0-cp39-cp39-win_amd64.whl is not a supported wheel on this platform.`

Odd, because I'm on a windows with ARM64 platform:
```C:\>echo %PROCESSOR_ARCHITECTURE%
AMD64```

Is this because of Python 3.11? Or I'm still missing something else?

### Issue Type

Build/Install

### Operating System

Windows 10

### Coral Device

M.2 Accelerator B+M

### Programming Language

Other

hjonnala, 2023-06-28:
> Is this because of Python 3.11? Or I'm still missing something else?

yes, the issue is with python 3.11. Please try with python3.9..

misosoup7, 2023-06-28:
Well I tried to install with WSL but now my computer wouldn't boot into windows.
I'll try removing WSL in safemode and see if that helps tonight.

It just spins forever right after bios.

misosoup7, 2023-07-02:
Recovered my windows install. But whenever I try to install Coral's drivers it'll cause the system to not boot up regardless of which Python version I try to use.

I will try moving it to a different M.2 Slot next time. Which may take a while as I'm planning to travel for a while.

Jo-Con-El, 2023-10-29:
> > Is this because of Python 3.11? Or I'm still missing something else?
> 
> yes, the issue is with python 3.11. Please try with python3.9..

What is different with Python 3.11 vs Python 3.9?

misosoup7, 2023-10-30:
Same issue with Python 3.9.2. Windows wouldn't let me install the drivers. As soon the drivers are installed, Windows refuses to boot. Only restoring to a prior restore point allows me to boot.