google-coral/pycoral #118: Unable to build on CentOS Stream 9

plysdyret, 2023-06-29:
### Description

Trying to build on CentOS Stream 9 bare metal server fails as root:

`
...
COMMIT coral-edgetpu-ubuntu-18.04
--> 3d1d0547343
Successfully tagged localhost/coral-edgetpu-ubuntu-18.04:latest
3d1d0547343ed555d2eabfe644bf4bd1017f84af02081101b315335a1aa00278
docker run --rm -i --tty -v /root/pycoral/:/workspace \
    "coral-edgetpu-ubuntu-18.04" /bin/bash -c "chmod a+w /; groupadd --gid 0 root; useradd -m -e '' -s /bin/bash --gid 0 --uid 0 root; echo 'root ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers; su root -c 'for cpu in k8 aarch64 armv7a; do make CPU=\${cpu} COMPILATION_MODE=opt -C /workspace/ pybind tflite wheel tflite-wheel || exit 1; done'"
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
groupadd: group 'root' already exists
useradd: user 'root' already exists
su: System error
make: *** [/root/pycoral/libcoral/docker/docker.mk:40: docker-build] Error 1
...
`

And fails as a regular user:

`
...
INFO: Build completed successfully, 708 total actions
mkdir -p /workspace/pycoral/pybind
mkdir: cannot create directory '/workspace/pycoral/pybind': Permission denied
...
`

### Issue Type

Build/Install

### Operating System

Linux

### Programming Language

Python 3.9

hjonnala, 2023-06-29:
> ... INFO: Build completed successfully, 708 total actions mkdir -p /workspace/pycoral/pybind mkdir: cannot create directory '/workspace/pycoral/pybind': Permission denied ...

Please add the sudo before dokcer command and try to build it: [pycoral/libcoral/docker/docker.mk:40:](https://github.com/google-coral/libcoral/blob/6589d0bb49c7fdbc4194ce178d06f8cdc7b5df60/docker/docker.mk#L40)

chrisruk, 2023-11-21:
@hjonnala Is there another solution to adding sudo, I tried that in Ubuntu 22.04, where I am getting a similar permission denied issue.

After prepending sudo I got - 

"docker: Error response from daemon: pull access denied for coral-edgetpu-ubuntu-18.04, repository does not exist or may require 'docker login': denied: requested access to the resource is denied."

chrisruk, 2023-11-22:
I managed to get around the permission denied issue doing the following -

```
diff --git a/docker/docker.mk b/docker/docker.mk
index 0609036..1483a68 100644
--- a/docker/docker.mk
+++ b/docker/docker.mk
@@ -32,11 +32,11 @@ docker-image:
            --build-arg IMAGE=$(DOCKER_IMAGE) $(DOCKER_MK_DIR)
 
 docker-shell: docker-image
-       docker run --rm -i --tty -v $(DOCKER_WORKSPACE):$(WORKSPACE) \
+       docker run --rm -i --tty -v $(DOCKER_WORKSPACE):$(WORKSPACE):z \
            --workdir $(WORKSPACE)/$(DOCKER_WORKSPACE_CD) \
-           $(DOCKER_TAG) /bin/bash -c "$(call run_command,$(DOCKER_SHELL_COMMAND))"
+           $(DOCKER_TAG) /bin/bash -c "$(call $(DOCKER_SHELL_COMMAND))"
 
 docker-build: docker-image
-       docker run --rm -i $(shell tty -s && echo --tty) -v $(DOCKER_WORKSPACE):$(WORKSPACE) \
-           $(DOCKER_TAG) /bin/bash -c "$(call run_command,$(MAKE_COMMAND))"
+       docker run --rm -i $(shell tty -s && echo --tty) -v $(DOCKER_WORKSPACE):$(WORKSPACE):z \
+           $(DOCKER_TAG) /bin/bash -c "$(call $(MAKE_COMMAND))"
```

Would be interested on how suitable this approach is.