google-coral/pycoral #125: AttributeError: 'Delegate' object has no attribute '_library'

zijam, 2023-10-21:
### Description

I have tried to run following command from the [getting started docs](https://coral.ai/docs/accelerator/get-started#3-run-a-model-on-the-edge-tpu) on my MacOS Ventura ARM64

```
 python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
```

But I get following error when running the command

```
~/workspace/coral/pycoral master* ❯ python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "/Users/siyalrach/workspace/coral/pycoral/examples/classify_image.py", line 139, in <module>
    main()
  File "/Users/siyalrach/workspace/coral/pycoral/examples/classify_image.py", line 87, in main
    interpreter = make_interpreter(*args.model.split("@"))
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/site-packages/pycoral-2.0.0-py3.9.egg/pycoral2/utils/edgetpu.py", line 87, in make_interpreter
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/site-packages/pycoral-2.0.0-py3.9.egg/pycoral2/utils/edgetpu.py", line 52, in load_edgetpu_delegate
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 89, in __init__
    self._library = ctypes.pydll.LoadLibrary(library)
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/ctypes/__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(libedgetpu.1.dylib, 0x0006): tried: 'libedgetpu.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibedgetpu.1.dylib' (no such file), '/Users/siyalrach/.pyenv/versions/3.9.16/lib/libedgetpu.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/siyalrach/.pyenv/versions/3.9.16/lib/libedgetpu.1.dylib' (no such file), '/opt/homebrew/lib/libedgetpu.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/libedgetpu.1.dylib' (no such file), '/Users/siyalrach/.pyenv/versions/3.9.16/lib/libedgetpu.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/siyalrach/.pyenv/versions/3.9.16/lib/libedgetpu.1.dylib' (no such file), '/opt/homebrew/lib/libedgetpu.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/libedgetpu.1.dylib' (no such file), '/usr/lib/libedgetpu.1.dylib' (no such file, not in dyld cache), 'libedgetpu.1.dylib' (no such file)
Exception ignored in: <function Delegate.__del__ at 0x1023a9820>
Traceback (most recent call last):
  File "/Users/siyalrach/.pyenv/versions/3.9.16/lib/python3.9/site-packages/tflite_runtime/interpreter.py", line 124, in __del__
    if self._library is not None:
AttributeError: 'Delegate' object has no attribute '_library'
```

### Relevant Log Output

```shell
python3 -c 'from pycoral.utils.edgetpu import get_runtime_version; print(get_runtime_version())'

BuildLabel(COMPILER=Apple LLVM 13.0.0 (clang-1300.0.29.30),DATE=redacted,TIME=redacted), RuntimeVersion(14)
```

zijam, 2023-10-21:
I got it working after running the following modification.

- When installing [Edge TPU runtime](https://coral.ai/docs/accelerator/get-started#runtime-on-mac) it generates a symlink to `/usr/local/lib/libedgetpu.1.dylib`:

```
~/workspace/coral/edgetpu_runtime ❯ sudo bash install.sh 
...
Using the maximum operating frequency for Coral USB devices.
==> Downloading https://formulae.brew.sh/api/formula.jws.json
#=#=-  #       #                                                                                                                                                                                                                                               
==> Downloading https://formulae.brew.sh/api/cask.jws.json

Warning: libusb 1.0.26 is already installed and up-to-date.
To reinstall 1.0.26, run:
  brew reinstall libusb
Installing Edge TPU runtime library [/usr/local/lib]...
Generating symlink [/usr/local/lib/libedgetpu.1.dylib]...
```
Which is not in the path, where`tflite_runtime` is looking in, after creating symlink `ln -s /usr/local/lib/libedgetpu.1.dylib /opt/homebrew/lib/libedgetpu.1.dylib` it worked.

- Also had to change `Image.ANTIALIAS` to `Image.LANCZOS` on line 95 in `examples/classify_image.py`, since **ANTIALIAS** was removed in [Pillow 10.0.0](https://pillow.readthedocs.io/en/stable/releasenotes/10.0.0.html#constants) and I had  version 10.1.0 on my system

```
~/workspace/coral/pycoral master* ❯ pip freeze                                                                                                                                                                   
numpy==1.26.1
Pillow==10.1.0
pycoral==2.0.0
tflite-runtime==2.5.0.post1
```

Moosems, 2023-12-25:
@zijam I still have this issue after creating the symlink, any ideas?

zijam, 2023-12-27:
@Moosems Can you send me some log info.