google-coral/pycoral #131: unable to build on rockylinux9

cjvirtucio87, 2023-11-06:
### Description

# Steps to reproduce

```console
git clone https://github.com/google-coral/pycoral "${cloned_pycoral}"
pushd "${cloned_pycoral}"
git submodule init
git submodule update
scripts/build.sh
```

# Expected

Completed build

# Actual

```
In file included from external/ruy/ruy/pack_arm.cc:16:
external/ruy/ruy/pack_arm.h:492:9: warning: multi-line comment [-Wcomment]
  492 | #endif  // (RUY_PLATFORM_NEON_64 || RUY_PLATFORM_NEON_32) && \
      |         ^
external/ruy/ruy/pack_arm.cc: In function 'void ruy::Pack8bitColMajorForNeon4Cols(const ruy::PackParams8bit&)':
external/ruy/ruy/pack_arm.cc:469:72: error: 'asm' operand has impossible constraints
  469 |         "q4", "q5", "q6", "q7", "q8", "q9", "q10", "q11", "q12", "q13");
      |                                                                        ^
Target //src:_pywrap_coral failed to build
INFO: Elapsed time: 25.202s, Critical Path: 13.18s
INFO: 240 processes: 114 internal, 126 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
make: *** [Makefile:152: pybind] Error 1
make: Leaving directory '/workspace'
make: *** [/tmp/tmp.K9R2WJyGOe/pycoral/libcoral/docker/docker.mk:40: docker-build] Error 1
```

# OS Information

```
[docker@cjvautomation ~]$ cat /etc/os-release
NAME="Rocky Linux"
VERSION="9.2 (Blue Onyx)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="9.2"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Rocky Linux 9.2 (Blue Onyx)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:9::baseos"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2032-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-9"
ROCKY_SUPPORT_PRODUCT_VERSION="9.2"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.2"
[docker@cjvautomation ~]$ uname -m
x86_64
[docker@cjvautomation ~]$ python3 --version
Python 3.9.16
```

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

### Relevant Log Output

```shell
In file included from external/ruy/ruy/pack_arm.cc:16:
external/ruy/ruy/pack_arm.h:492:9: warning: multi-line comment [-Wcomment]
  492 | #endif  // (RUY_PLATFORM_NEON_64 || RUY_PLATFORM_NEON_32) && \
      |         ^
external/ruy/ruy/pack_arm.cc: In function 'void ruy::Pack8bitColMajorForNeon4Cols(const ruy::PackParams8bit&)':
external/ruy/ruy/pack_arm.cc:469:72: error: 'asm' operand has impossible constraints
  469 |         "q4", "q5", "q6", "q7", "q8", "q9", "q10", "q11", "q12", "q13");
      |                                                                        ^
Target //src:_pywrap_coral failed to build
INFO: Elapsed time: 25.202s, Critical Path: 13.18s
INFO: 240 processes: 114 internal, 126 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
make: *** [Makefile:152: pybind] Error 1
make: Leaving directory '/workspace'
make: *** [/tmp/tmp.K9R2WJyGOe/pycoral/libcoral/docker/docker.mk:40: docker-build] Error 1
```
```

tranzmatt, 2023-11-08:
I'm experiencing the same issue on Ubuntu 22.04, but I'm assuming since it's running inside Docker, it shouldn't matter?

cjvirtucio87, 2023-11-08:
Looking at the error, it seems to lie with an external dependency? I'm not seeing `ruy` in any of the coral repos. So maybe it has to do with one of the non-coral ones [here](https://github.com/google-coral/pycoral/blob/ae743b0b234d66d784826ccfbb4607a69dc1a2a8/src/BUILD#L83-L100).

tranzmatt, 2023-11-09:
It's limited to the arm7f architecture, so if you don't need it, just comment it out of the build script.