google-coral/pycoral #133: Unable to Install Pycoral on Raspberry Pi 5

tribui141108, 2023-11-12:
### Description

The latest version of raspbian os for raspberry pi 5 currently only supports Python 3.11, not Python 3.10 or newer. But for Pycoral to be installed, it requires Python 3.10 or newer. Is there any way that you can support Raspberry Pi 5?

This is the command I used to run:
```bash
sudo apt-get install python3-pycoral
```

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

USB Accelerator

### Programming Language

Other

### Relevant Log Output

```shell
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-tflite-runtime : Depends: python3 (< 3.10) but 3.11.2-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

HRA42, 2023-12-04:
same problem

Jo59Devs, 2023-12-06:
same problem

sjunger, 2023-12-13:
same problem

sjunger, 2023-12-13:
tflite-runtime can be installed in my venv wirh "pip install tflite-runtime".

Now when I run "pip install pycoral" I get the following messages:

(.venv) ┌─[ ]-[17:45:30]-[sjunger@xxxxxxxxxxxxx]-[~/Documents/xxxxxxxxxxxxxx]
└─$ pip install pycoral
Collecting pycoral
  Using cached pycoral-0.2.0-py3-none-any.whl.metadata (9.0 kB)
Collecting requests>=2.23.1 (from pycoral)
  Using cached requests-2.31.0-py3-none-any.whl.metadata (4.6 kB)
Collecting progressbar2>=3.53.1 (from pycoral)
  Using cached progressbar2-4.2.0-py2.py3-none-any.whl (27 kB)
Collecting area>=1.1.1 (from pycoral)
  Using cached area-1.1.1.tar.gz (4.1 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting tenacity>=8.0.1 (from pycoral)
  Using cached tenacity-8.2.3-py3-none-any.whl.metadata (1.0 kB)
Collecting pandas>=2.1.1 (from pycoral)
  Using cached pandas-2.1.4-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (18 kB)
Collecting pyproj>=1.9.5.1 (from pycoral)
  Using cached pyproj-3.6.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (31 kB)
Collecting shapely>=1.7.1 (from pycoral)
  Using cached shapely-2.0.2-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (7.0 kB)
Collecting fiona>=1.8.20 (from pycoral)
  Using cached fiona-1.9.5.tar.gz (409 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [3 lines of output]
      <string>:86: DeprecationWarning: The 'warn' function is deprecated, use 'warning' instead
      WARNING:root:Failed to get options via gdal-config: [Errno 2] No such file or directory: 'gdal-config'
      CRITICAL:root:A GDAL API version must be specified. Provide a path to gdal-config using a GDAL_CONFIG environment variable or use a GDAL_VERSION environment variable.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

---

Python 3.11.6 (main, Oct  8 2023, 05:06:43) [GCC 13.2.0] on linux

sjunger, 2023-12-13:
Okay, forget "tflite-runtime". This version would be too new anyway. pycoral requires 2.5.0.post1 and I now have 2.14.0

HRA42, 2023-12-13:
I just run pycoral in a docker container for now.. performance is good enoght for what I need to do.

Jo59Devs, 2023-12-14:
> I just run pycoral in a docker container for now.. performance is good enoght for what I need to do.

which docker image can you recommend?
Thanks!

HRA42, 2023-12-14:
As the documentation requests, I used Debian 10

nathan-stewart, 2023-12-27:
Pycorlal is basically abandoned.  Using pyenv and python-3.9.18 I have gotten https://github.com/oberluz/pycoral fork to work. At least, after fixing a couple s/PIL.Image.ANTIALIAS/PIL.Image.LANCZOS/ breakages (the PIL version this pulls has removed ANTIALIAS), I can run the classify and detect_image examples to run. I haven't yet managed to get it to work with the model I'm trying (tf2_ssd_mobilenet_v2_coco17_ptq_edgetpu.tflite)

feranick, 2024-03-04:
Support for EdgeTPu is coming back. Updated libcoral and pycoral libraries for Python 3.11+ and modern TF are coming. The libedgetpu official repo was just updated based on my PR, and is compatible with TF 2.15.0

I am working to get libcoral/pycoral merged to the stable repo by the time TF 2.16.0 reaches stable.

Details here:

https://github.com/google-coral/pycoral/issues/137#issuecomment-1974048543