google-coral/pycoral #137: Update pycoral for Python 3.11 and modern versions of Tensorflow

feranick, 2023-12-13:
### Description

Pycoral is available only with support for Python 3.9. Most distribution now no longer ship with that version, rather either 3.10 or 3.11. So there is factually no version of pycoral available to download. Furthermore, building pycoral from source is not compatible with Python 3.10 or 3.11 (again, not supported) and relies on an ancient version of tensorFlow (2.5.0) to work. While this al points to a lack of maintenance, it would be great to at least update so it can be used with, well, pretty much any modern version of OS.

### Issue Type

Build/Install

### Operating System

Linux, Ubuntu, Mac OS

### Coral Device

USB Accelerator, M.2 Accelerator A+E, M.2 Accelerator B+M, M.2 Accelerator with dual Edge TPU

### Other Devices

Raspberry Pi 3, Rapsberry Pi 4

### Programming Language

Other

tranzmatt, 2024-01-23:
Has this project been officially abandoned?

feranick, 2024-01-25:
There hasn't been any comment, action, or even statement in a very long time. the repos are frozen in time. Such a pity when every chip manufacturer is presenting NPUs as the next big thing, and Google (who pioneered it with the edgeTPU so long ago) decided to drop the ball. To make it worse, no official statement either, and a community that relies/invested on it is left it to its own devices (pun very much intended). And to make it worse, coral.ai still sells these products as fully supported when they are nothing but.

Sad, and infuriating.

feenadesign, 2024-02-16:
I'm experiencing the same issue on 64 bit Raspberry Pi OS 12.4 running on a Pi 4 which has python 3.11.2.  I tried to build pycoral myself and have not managed to make things work yet.

I guess all the layoffs have hit this team pretty hard.

feranick, 2024-02-16:
I ended up forking the `libedgetpu` and bringing it up to date with compatibility to the latest version of TF. This is all I really needed (as I don't use `pycoral`, but directly access the edgeTPU with`tflite_runtime`) so I can fully use the edgeTPU. I took a look at `pycoral` but it seems that a whole lot more work is needed to bringing it up to date. 

https://github.com/feranick/libedgetpu

Emojibotbot, 2024-02-16:
> I ended up forking the `libedgetpu` and bringing it up to date with compatibility to the latest version of TF. This is all I really needed (as I don't use `pycoral`, but directly access the edgeTPU [...]

I was able to install the edgetpu runtime without issues on my rig (but not have tried to run it in anyway yet.)  What made you need to update?

feranick, 2024-02-16:
> I was able to install the edgetpu runtime without issues on my rig (but not have tried to run it in anyway yet.) What made you need to update?

That is the point. The provided `whl` install fine if you have he correct Phython environment (see below), but they segfault if you are running `tlite_runtime` > 2.11.1. 

The wheels provided are only available for up to Python 3.9, su pretty useless in any modern Linux release (Raspbian/Debian-bookworm current uses 3.11, ubuntu 22.04 uses 3.10). It may work with "legacy" debian-bullseye which uses Python 3.9.

feranick, 2024-02-16:
Plenty of discussion here:

https://github.com/google-coral/edgetpu/issues/820#issuecomment-1921289528

feranick, 2024-02-16:
There is also another issue: the google-dkms gasket is incompatible with linux kernel >= 6.5 and therefore the edgeTPU won't be recognized. The solution is a recompile of the gasket itself, following instructions here:

https://github.com/google-coral/edgetpu/issues/820

Emojibotbot, 2024-02-16:
I see you have done a lot of work to update things already & nice job!  I thought the issue was just that pycoral was not up to date, but the lack of updates much more systemic across all of google-coral.

Emojibotbot, 2024-02-16:
Is it safe to assume your fork would not work?  https://github.com/feranick/pycoral

feranick, 2024-02-16:
Correct. If it will ever work, I will post binaries there, but I have 
limited time to spend on it at the moment.

Thanks

--------------uAcACjLxWMBrvKW4lt9XPhZx
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    Correct. If it will ever work, I will post binaries there, but I
    have limited time to spend on it at the moment. <br>
    <br>
    Thanks<br>
    <br>
    <div class="moz-cite-prefix">On 2/16/24 12:21 PM, Emojibotbot wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Is it safe to assume your fork would not work? <a href="https://github.com/feranick/pycoral" moz-do-not-send="true" class="moz-txt-link-freetext">https://github.com/feranick/pycoral</a></p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;google-coral/pycoral/issues/137/1948939758</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/google-coral/pycoral/issues/137#issuecomment-1948939758",
"url": "https://github.com/google-coral/pycoral/issues/137#issuecomment-1948939758",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------uAcACjLxWMBrvKW4lt9XPhZx--

feranick, 2024-02-29:
Thanks to @pkgoogle and @namburger, the libdgetpu repo/library has been updated, a big step forward. Next that should be updated is libcoral/pycoral, the frontend libraries that are (supposedly) the standard way to interact with the edgeTPU. My fork still does not compile, and several issues are at play. There is an open issue/PR for TF that is supposed to fix one of the issues: [Change visibility status for lite/schema/schema_conversion_utils to allow compilation of external libraries](https://github.com/tensorflow/tensorflow/issues/63074). If there is a way for this PR to be merged, that'd be great, since in the current state, such visibility issue in TF (which is downloaded by Bazel in tmp_folder during the compilation of libcoral) hangs compilation of pycoral.

feranick, 2024-03-01:
I finally managed to successfully compile libcoral through this [new fork of mine](https://github.com/feranick/libcoral).

I submitted a PR request: https://github.com/google-coral/libcoral/pull/36

Of course, this needs the [visibility issue](https://github.com/tensorflow/tensorflow/pull/63075) to be fixed first. Yet, compilation can be successfully done manually, until the issue above is fixed. Two steps:

1. Change last line in WORKSPACE from:
```
path = "/usr/system_libs",
```
to
```
path = "/",
```
as the former is designed when operating with Docker.

2. Edit: `nano ~/.cache/bazel/_bazel_XXXX/XXXXXXX/external/org_tensorflow/tensorflow/lite/schema/BUILD`
and remove `visibility = [":utils_friends"],`

from both:
```
cc_library(
    name = "schema_utils",
    srcs = ["schema_utils.cc"],
    hdrs = ["schema_utils.h"],
    compatible_with = get_compatible_with_portable(),
    visibility = [":utils_friends"],
    deps = [
        ":schema_fbs",
        "//tensorflow/lite/kernels/internal:compatibility",
        "@flatbuffers//:runtime_cc",
    ],
)

cc_library(
    name = "schema_conversion_utils",
    srcs = ["schema_conversion_utils.cc"],
    hdrs = ["schema_conversion_utils.h"],
    compatible_with = get_compatible_with_portable(),
    visibility = [":utils_friends"],
    deps = [
        ":schema_fbs",
        "//tensorflow/lite/kernels/internal:compatibility",
        "@flatbuffers",
    ],
)
```

feranick, 2024-03-01:
[A new fork for `pycoral` is now available](https://github.com/feranick/pycoral), which relies on the [`libedgetpu`](https://github.com/feranick/libedgetpu) and [`libcoral`](https://github.com/feranick/libcoral) forks, and it has been tested to work. All three libraries are currently targeting the upcoming TensorFlow version 2.16.0 (currently in rc0) status, with the hope that the [visibility issue](https://github.com/tensorflow/tensorflow/pull/63075) is fixed in that release. Once TF 2.16.0 is officially released as stable, I will create a PR for the updated `libedgetpu` (which now targets TF2.15.0 in the official repo). Upon merging `libedgetpu` into the official repo (which currently targets TF 2.15.0), [another PR is ready](https://github.com/google-coral/libcoral/pull/36) for the updated `libcoral`. Once that is merged, you guessed it, I will push a PR for `pycoral` based [on this fork](https://github.com/feranick/pycoral).

In the meantime, one can already build `libedgetpu` from the [official repo](https://github.com/google-coral/libedgetpu) (against TF 2.15.0) [or the fork ](https://github.com/feranick/libedgetpu) needed for this upcoming version of pycoral. With that, `pycoral` can be built as instructed in the [README file in the forked repo](https://github.com/feranick/pycoral), with the only caveat that the visibility issue needs to be manually addressed, as detailed [in the comment above.](https://github.com/google-coral/pycoral/issues/137#issuecomment-1972454039).

Thanks again to @namburger and @pkgoogle for pushing forward the effort to modernize the EdgeTPU libraries.

feranick, 2024-03-01:
> [A new fork for `pycoral` is now available](https://github.com/feranick/pycoral), which relies on the [`libedgetpu`](https://github.com/feranick/libedgetpu) and [...]

feranick, 2024-03-22:
Which version of libedgetpu are you using?

feranick, 2024-03-22:
Can you provide a usable script to try this out? Better yet, can you run one of the official examples available here? https://coral.ai/docs/edgetpu/tflite-python/#inferencing-example

feranick, 2024-03-22:
Note that with pycoral, the tflite model is used directly (i.e. there is no need to convert it into _edgetpu.tflite), so it is possible that the runtime is pretty much the same. If you can, you should try to run the tflite and _egetpu.tflite independently without using pycoral, but directly from tensorflow.

feranick, 2024-03-22:
Also keep in mind that the fork is still experimental due to compatibility issues in TF 2.16.1.

feranick, 2024-03-22:
Thank you for trying. Using libedgetpu directly works smoothly and with no problems. As pycoral goes, a major issue will be resolved when TF 2.17.0 will be released, that is preventing compilation of pycoral. There are also other issues at play, especially w.r.t. armhf. In all honesty, at the moment, the pycoral is a hackjob because of this inconsistencies of so many different elements in needs to play along (TF, libedgetpu, libcoral, pycoral)... But we'll get there.

feranick, 2024-03-24:
Great. Again, I'll keep working on pycoral, but that will take time. libedgetpu seems to be remaining well aligned with TF, so direct access to edgetpu through it should be the stable choice, for now.

julianmata, 2024-04-26:
Hey @feranick, I am trying to install your pycoral build, I followed the guide here:
https://github.com/feranick/pycoral/releases
When I try with python version of 3.11 it works okay but when I try 3.10 builds I am getting this error:
`Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycoral.utils import edgetpu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/julian/.local/lib/python3.10/site-packages/pycoral/utils/edgetpu.py", line 24, in <module>
    from pycoral.pybind._pywrap_coral import GetRuntimeVersion as get_runtime_version
ImportError: Python version mismatch: module was compiled for Python 3.11, but the interpreter version is incompatible: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0].`

Just to be correct I installed the driver like this:
`wget https://github.com/feranick/libedgetpu/releases/download/16.0TF2.16.1-1/libedgetpu1-std_16.0tf2.16.1-1.ubuntu22.04_amd64.deb`
`sudo dpkg -i libedgetpu1-std_16.0tf2.16.1-1.ubuntu22.04_amd64.deb `
then installed runtime wheel like this:
`pip3 install https://github.com/feranick/TFlite-builds/releases/download/v.2.16.1/tflite_runtime-2.16.1-cp310-cp310-linux_x86_64.whl`
then installed pycoral wheel like this:
`pip3 install https://github.com/feranick/pycoral/releases/download/v2.0.1TF2.16.1/pycoral-2.0.1-cp310-cp310-linux_x86_64.whl`
after that I tried to use it as I have shown above, where I try to import edgetpu and I am getting that error.
Am I missing something, and as a bonus can you provide your env setup for building the wheel?

Thank you, for any answer I would appreciate it!

feranick, 2024-04-28:
PyCoral is currently not stable, until tflite_runtime reaches v 2.17.0 (with the next release of Tensorflow). You should wait until then.

EnziinSystem, 2024-05-08:
@feranick

You are the hope for the Edge TPU community after Google showed irresponsibility by abandoning the project.

We have the right to ask because we bought the hardware from Google.

Currently, I have the following:

- * Raspberry Pi-5 with OS 64-bit, Kernel 6.6.30-v8+
- * AI Coral Edge TPU module from Pinboards
- * PiCamera module V3
- * Python 3.11

What steps do I need to take to be able to use TPU with Python 3.11 and avoid the maze of version conflicts?

Thank you.

han-so1omon, 2024-05-08:
@EnziinSystem Try the steps [here](https://www.jeffgeerling.com/blog/2023/pcie-coral-tpu-finally-works-on-raspberry-pi-5) if any issues arise. I had some issues with the Hat AI from Pineboards

EnziinSystem, 2024-05-09:
@feranick 

I followed your instructions and the results were as follows:

```
$ uname -a
Linux smarthome 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

$ lspci -nn | grep 089a
0000:03:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]

$ ls -la /dev/apex_0
crw-rw---- 1 root apex 120, 0 May  9 15:03 /dev/apex_0

$ sudo lspci -vvv|grep -i MSI-X
        Capabilities: [d0] MSI-X: Enable- Count=128 Masked-
        Capabilities: [d0] MSI-X: Enable+ Count=9 Masked-
        Capabilities: [b0] MSI-X: Enable+ Count=61 Masked-
```
Because I cannot use kernel 6.6.30 , I recompile DTB with msi-parent = <0x6d>

Run demo

```
$ python3 test_libedgetpu.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 166, in load_delegate
    delegate = Delegate(library, options)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 104, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/developer/coral/test_edgetpu/test_libedgetpu.py", line 12, in <module>
    model = tflite.Interpreter(model_path='model_classifier_CNN_edgetpu.tflite', experimental_delegates=[tflite.load_delegate(edgeTPUSharedLib,{})])
                                                                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 168, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```

Although there are still errors, I still thank you for your help.

EnziinSystem, 2024-05-09:
> @EnziinSystem I think that might have to do something with your specific setup. I run a Pi4 (Bookworm) and Coral over USB and using libedgetpu and the tflite runtime (without pycoral) worked literally on the first try.

If I run via Docker and Python 3.9, I also run it from the first time.
The problem here is that Python 3.11 and the latest Kernels 6.6.30 and TPU cards from PCI have compatibility issues.

feranick, 2024-05-09:
The issue is with the PCIe HAT. `gaster-dkms` has support for kernels >6.8 and Python 3.11 is proven to work fine with the binaries provided. I'd recommend reaching out for the maker of the HAT as it is clearly an issue they should be provide support for. Again, I used the USB edgeTPU with Python 3.11 and kernel 6.6.28 (the officially supported one, any other from rpi-update is not recommended for stability and daily usage as recommended by rpi-update itself) and the M.2 version of the edgeTPU on Dektop Debian running a similar setup.

EnziinSystem, 2024-05-10:
After following the instructions at [Pinboards](https://pineboards.io/blogs/tutorials/how-to-configure-the-google-coral-edge-tpu-on-the-raspberry-pi-5)

I can use TPU M2 card on Kernel 6.6.30 with **dtoverlay=pineboards-hat-ai** instead of recompiling DTB and change **ms-parent**

```
$ uname -a
Linux smarthome 6.6.30-v8+ #1762 SMP PREEMPT Thu May  9 15:17:51 BST 2024 aarch64 GNU/Linux
```

Install TFlite - Update libedgetpu1 from @feranick  repos

```
wget https://github.com/feranick/libedgetpu/releases/download/16.0TF2.16.1-1/libedgetpu1-std_16.0tf2.16.1-1.bookworm_arm64.deb

sudo dpkg -i libedgetpu1-std_16.0tf2.16.1-1.bookworm_arm64.deb

wget https://github.com/feranick/TFlite-builds/releases/download/v.2.16.1/tflite_runtime-2.16.1-cp311-cp311-linux_aarch64.whl

sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.bkp

sudo pip3 install --upgrade tflite_runtime-2.16.1-cp311-cp311-linux_aarch64.whl
```

Run a demo

https://github.com/google-coral/tflite/tree/master/python/examples/classification

```
$ python3 classify_image.py \
  --model models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
  --labels models/inat_bird_labels.txt \
  --input images/parrot.jpg
/home/developer/coral/google-coral/tflite/python/examples/classification/classify_image.py:103: DeprecationWarning: ANTIALIAS is deprecated and will be removed in Pillow 10 (2023-07-01). Use LANCZOS or Resampling.LANCZOS instead.
  image = Image.open(args.input).convert('RGB').resize(size, Image.ANTIALIAS)
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
12.5ms
2.3ms
2.3ms
2.3ms
2.3ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.77734
```

Note: We still can't use Pycoral like @feranick said.

EnziinSystem, 2024-05-13:
@feranick 

Today 05/13/2024, TensorFlow Lite released 2.17 and I can build it successfully on Pi-5.

```
pip3 install pybind11

git clone https://github.com/tensorflow/tensorflow.git

PYTHON=python3 tensorflow/tensorflow/lite/tools/pip_package/build_pip_package_with_cmake.sh native
```

Since you mentioned this before, what improvements or problems does tflite-2.17 have for pycoral?

feranick, 2024-05-13:
Tensorflow (including lite) v2.17 has not been releases, as it is not showing in the official releases. 

https://github.com/tensorflow/tensorflow/releases

Usually google first does 1-3 releases candidates, which are also not showing. What you have is a dev tree that may lead at some point to 2.17. But is NOT 2.17 stable. 

Once 2.17 is out, pycoral can finally be compiled for it in a way that it is compatible with tensorflow. It's minor detail due to an internal library. There are of course other issues, but until 2.17 is actually out nothing happens.

feranick, 2024-05-13:
If you are interested in the details follow here:

https://github.com/google-coral/libcoral/pull/36

and here:

https://github.com/google-coral/libedgetpu/pull/60

feranick, 2024-05-13:
And of course, if you want to use the 2.17 dev branch, you can compile libedgetpu yourself against it, and possibly attempt pycoral as well as indicated in my fork:

https://github.com/feranick/pycoral

But I don't provide any support for it (as it might crash and burn). Once 2.17 I will start again trying to merge pycoral into upstream with all changes that are needed.

withanhdammit, 2024-05-19:
I'm running a desktop version of Debian 12 using the 6.6.13 kernel from bookworm-backports
```
uname -a
Linux frigate 6.6.13+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1~bpo12+1 (2024-02-15) x86_64 GNU/Linux
```
My Coral TPU is this one [](https://www.amazon.com/dp/B0CY231Q61), and my computer does recognize it's installed (sadly only one of the cores, but that's OK)
```
lspci | grep Coral
06:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU
```
I modified your steps slightly to use the amd64/x86_64 versions in your repo
```
mkdir ~/edgetpu-inst
cd edgetpu-inst
wget https://github.com/feranick/gasket-driver/releases/download/v1.0-18.1/gasket-dkms_1.0-18_all.deb
sudo apt install linux-headers-generic dkms linux-headers-6.1.0-21-amd64 linux-headers-6.1.0-21-common linux-headers-amd64 linux-kbuild-6.1
sudo dpkg -i gasket-dkms_1.0-18_all.deb
wget https://github.com/feranick/libedgetpu/releases/download/16.0TF2.16.1-1/libedgetpu1-std_16.0tf2.16.1-1.bookworm_amd64.deb
sudo dpkg -i libedgetpu1-std_16.0tf2.16.1-1.bookworm_amd64.deb
wget https://github.com/feranick/TFlite-builds/releases/download/v.2.16.1/tflite_runtime-2.16.1-cp311-cp311-linux_x86_64.whl
sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.bkp
sudo pip3 install --upgrade tflite_runtime-2.16.1-cp311-cp311-linux_x86_64.whl
```
when I run the 'test_libedgetpu.py' I get:
```
~/edgetpu-inst/test_edgetpu$ python3 test_libedgetpu.py 
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 166, in load_delegate
    delegate = Delegate(library, options)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 104, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/darhl/edgetpu-inst/test_edgetpu/test_libedgetpu.py", line 12, in <module>
    model = tflite.Interpreter(model_path='model_classifier_CNN_edgetpu.tflite', experimental_delegates=[tflite.load_delegate(edgeTPUSharedLib,{})])
                                                                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tflite_runtime/interpreter.py", line 168, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from libedgetpu.so.1
```
At this point, I'm a little over my head, any suggestions for where to look to resolve this?

EnziinSystem, 2024-05-20:
@withanhdammit 

You should compile the drive from source instead of installing from the deb file.

```
git clone https://github.com/google/gasket-driver.git

cd gasket-driver

sudo debuild -us -uc -tc -b

Go back to the parent directory and install the built package:

cd ..

sudo dpkg -i gasket-dkms_1.0-18_all.deb

```

The installation process also requires linux-header files.

Camilochiang, 2024-05-22:
Hei @EnziinSystem . I have a situation with a pineberry hat (the old version, from january). I can run my models in models but randomly the TPU hang and freeze after reading it. Have you experience something similar? Thanks!

EnziinSystem, 2024-05-22:
> Hei @EnziinSystem . I have a situation with a pineberry hat (the old version, from january). I can run my models in models but randomly the TPU hang and freeze after reading it. Have you experience something similar? Thanks!

After using [HAT AI Coral](https://pineboards.io/products/hatdrive-ai-coral-edge-tpu-bundle-nvme-2230-2242-gen-2-for-raspberry-pi-5) and installing the operating system on the SSD, I ran a simple example of the Picamera2 library, I also experienced that sometimes the system paused for a while and then continued running. 

https://github.com/raspberrypi/picamera2/blob/main/examples/tensorflow/real_time_with_labels.py

This example does not use TPU but only inferences based on Tensoflow Lite.

I don't know the reason why.

withanhdammit, 2024-06-04:
> @withanhdammit
> 
> You should compile the driver from source instead of installing from the deb file.
> 
> ```
> git clone https://github.com/google/gasket-driver.git
> 
> cd gasket-driver
> 
> sudo debuild -us -uc -tc -b
> 
> Go back to the parent directory and install the built package:
> 
> cd ..
> 
> sudo dpkg -i gasket-dkms_1.0-18_all.deb
> ```
> 
> The installation process also requires linux-header files.

@EnziinSystem, I updated my original post, but thought I'd add the reply here, too.  I was running in a user context, but when I ran it with sudo, it worked perfectly.  It's properly running on my Frigate NVR now.

hillzzz, 2024-06-12:
> @feranick
>
> Today 05/13/2024, TensorFlow Lite released 2.17 and I can build it successfully on Pi-5.
>
> ```
> pip3 install pybind11
>
> git clone https://github.com/tensorflow/tensorflow.git
>
> PYTHON=python3 [...]

I can confirm a successful build but I followed @feranick 's very helpful instructions and the build no from RELEASE.md is # 

**Release 2.18.0**
## TensorFlow
I am running the Dual ai Pineberry hat with the Dual EdgeTPU card. Python 3.11.2, Kernel Linux Raspian 6.6.32-v8+ #1769 SMP PREEMPT Thu Jun  6 11:38:54 BST 2024 aarch64 GNU/Linux

feranick, 2024-06-13:
I'd note, once again, that the current stable version of TF (standard and lite) is 2.16.1. You may see more recent versions compiled IF, and only IF you compile from the master head of the git repo, which also means that you are running a very unstable version. If you want to run stable, you should get the latest stable version from here: 

https://github.com/tensorflow/tensorflow/releases

Note: Google usually releases "RC" versions of the next stable in the same page, for testing. As for myself, I may do that but ultimately only publish builds against stable.

feranick, 2024-06-21:
New test builds for the recently released TF 2.17.0-rc0. Note this is a release candidate, not stable, yet.
Builds for `tflite_runtime`:
https://github.com/feranick/TFlite-builds/releases/tag/v2.17.0-rc0
Builds for `libedgetpu`:
https://github.com/feranick/libedgetpu/releases/tag/16.0TF2.17.0rc0-1
Builds for `pycoral`:
https://github.com/feranick/pycoral/releases/tag/v2.0.2TF2.17.0rc0

Note: I have not fully test them, so use at your own risk.

Alvarofg, 2024-07-16:
[TensorFlow 2.17.0](https://github.com/tensorflow/tensorflow/releases/tag/v2.17.0) is out now. I hope we see some activity here soon :)

feranick, 2024-07-16:
New builds for the recently released TF 2.17.0. 
Builds for tflite_runtime:
https://github.com/feranick/TFlite-builds/releases/tag/v2.17.0
Builds for libedgetpu:
https://github.com/feranick/libedgetpu/releases/tag/16.0TF2.17.0-1
Builds for pycoral:
https://github.com/feranick/pycoral/releases/tag/2.0.2TF2.17.0

Note: If you don't see your platform, it means that at this stage iit is not supported.

PederLAB, 2024-07-24:
Thank you @feranick for your long time work, to extend the value of Coral (USB) TPU!

virnik0, 2024-08-05:
> New test builds for the recently released TF 2.17.0-rc0. Note this is a release candidate, not stable, yet. Builds for `tflite_runtime`: https://github.com/feranick/TFlite-builds/releases/tag/v2.17.0-rc0 Builds for `libedgetpu`: https://github.com/feranick/libedgetpu/releases/tag/16.0TF2.17.0rc0-1 Builds for `pycoral`: https://github.com/feranick/pycoral/releases/tag/v2.0.2TF2.17.0rc0
> 
> Note: I have not fully test them, so use at your own risk. Note 2: If you don't see your platform, it means that at this stage iit is not supported.

Would it be possible to build all three files for python3.12? I am building it on my own rig, but so far, facing various issues. For one, pycoral built for python3.12 doesn't recognize tflow which is already installed and built using your tree. pip just says that it looks for version (13.2?) of tflow, which is already installed, but gets back 'none'

feranick, 2024-08-05:
I can't yet compile on Python3.12, as I am getting errors related to cross-compilation with arm platforms. I a,so don't run yet on ubuntu:24.04. 

Are you compiling directly? Are you doing via docker? The former should work fine on amd64.

virnik0, 2024-08-05:
> I can't yet compile on Python3.12, as I am getting errors related to cross-compilation with arm platforms. And I can't compile directly yet as I don't run on ubuntu:24.04.
> 
> Are you compiling directly? Are you doing via docker? The former should work fine on amd64.

I tried both approaches. Direct one was partially successful. It produced wheel, but that wheel wasn't usable. I borked my python install after deciding to use snake's python3.11 build, which in turn required older numpy, updates to grub generator sripts, etc....

I am glad I am using ZFS, so the major issue for me was to select correct snapshot to restore system state from.

I am on native platform, currently x86_64 (but I also have amd64 here, which uses older python3.11 and was installed about five years ago along with coraledgetpu, but I wouldn't like to change anything there as that is prod system).

Once I restore my system state back to python3.12 env, I am open to try anything which would allow me to use pycoral on python3.12.

elitistphoenix, 2024-08-07:
This is the error I get at the end if that helps

`ERROR: /home/user/.cache/bazel/_bazel_user/eab0d61a99b6696edb3d2aff87b585e8/external/XNNPACK/BUILD.bazel:1869:19: Compiling src/amalgam/gen/avxvnni.c failed: (Exit 1): x86_64-linux-gnu-gcc failed: error executing command (from target @XNNPACK//:avxvnni_prod_microkernels)
  (cd /home/user/.cache/bazel/_bazel_user/eab0d61a99b6696edb3d2aff87b585e8/sandbox/processwrapper-sandbox/281/execroot/pycoral && \
  exec env - \
    PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/x86_64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -msse4.2 -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-std=c99' -MD -MF bazel-out/k8-opt/bin/external/XNNPACK/_objs/avxvnni_prod_microkernels/avxvnni.d '-frandom-seed=bazel-out/k8-opt/bin/external/XNNPACK/_objs/avxvnni_prod_microkernels/avxvnni.o' '-DBAZEL_CURRENT_REPOSITORY="XNNPACK"' -iquote external/XNNPACK -iquote bazel-out/k8-opt/bin/external/XNNPACK -isystem external/XNNPACK/include -isystem bazel-out/k8-opt/bin/external/XNNPACK/include -isystem external/XNNPACK/src -isystem bazel-out/k8-opt/bin/external/XNNPACK/src '-DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION' -Iinclude -Isrc -mavx2 -mavxvnni -mf16c -mfma '-std=c99' -O2 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/XNNPACK/src/amalgam/gen/avxvnni.c -o bazel-out/k8-opt/bin/external/XNNPACK/_objs/avxvnni_prod_microkernels/avxvnni.o)

Configuration: 0a92d423cd1a3f2c7226e22abb0a87ec8b331af77c99153b2f1e43d8d5515f84
Execution platform: @local_execution_config_platform//:platform

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
x86_64-linux-gnu-gcc: error: unrecognized command line option '-mavxvnni'; did you mean '-mavx512vnni'?
Target //src:_pywrap_coral failed to build
INFO: Elapsed time: 88.833s, Critical Path: 26.39s
INFO: 561 processes: 292 internal, 269 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
make: *** [Makefile:152: pybind] Error 1
make: Leaving directory '/workspace'
make: *** [/mnt/storage/pycoral/libcoral/docker/docker.mk:40: docker-build] Error 1`

feranick, 2024-08-07:
Which system are you trying to compile it for? From the error it seems Ubuntu 22.04. The compilation fails because gcc is no longer supported by TF 2.17.0 and the version of clang in Ubuntu 22.04 is too old (and that flag is not supported). 

Details here:
https://github.com/tensorflow/tensorflow/issues/70199#issuecomment-2189084976

You might want to use docker builds, as suggested.

virnik0, 2024-08-07:
I am on Ubuntu 24.04. I'll try with clang18 (was using gcc till now). I don't mind building all three components myself, as I did so for prior version on python3.10, but this time I am on python 3.12 and it still fails. Docker builds are also failing (I do have docker installed, I know how to use it, but for the benefit of it, I resort to scripts/build.sh script to do the lifting.

feranick, 2024-08-07:
Clang 18 *should* work. Again the error you see is specific to GCC that is no longer supported in building anything related to TF.

virnik0, 2024-08-07:
Well, my problem actuall isn't clang or gcc, but rather bazel. If I'll use bazel version 4.5.0, it sorta works. If I'll use bazel 6.x, I don't get too far:

```
root@wvojtas1:~/repositories/coral/pycoral_faranick# bazel clean --expunge
INFO: Starting clean (this may take a while). Consider using --async if the clean takes more than several minutes.
root@wvojtas1:~/repositories/coral/pycoral_faranick# TF_PYTHON_VERSION=3.12 make
Starting local Bazel server and connecting to it...
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
Loading: 0 packages loaded
PYTHON_BIN_PATH=/usr/bin/python3 bazel build --compilation_mode=opt --copt=-DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION --cpu=k8 --linkopt=-L/root/repositories/coral/pycoral_faranick/libedgetpu_bin/direct/k8 --linkopt=-l:libedgetpu.so.1 --linkopt=-Wl,--strip-all \
    --embed_label='TENSORFLOW_COMMIT=ad6d8cc177d0c868982e39e0823d0efbfb95f04c' \
    --stamp \
    //src:_pywrap_coral
ERROR: /root/repositories/coral/pycoral_faranick/WORKSPACE:154:21: fetching new_local_repository rule //external:glog: /root/repositories/coral/pycoral_faranick/external/libcoral/third_party/glog/BUILD is not a regular file; if you're using a relative or absolute path for `build_file` in your `new_local_repository` rule, please switch to using a label instead
ERROR: no such package '@@glog//': /root/repositories/coral/pycoral_faranick/external/libcoral/third_party/glog/BUILD is not a regular file; if you're using a relative or absolute path for `build_file` in your `new_local_repository` rule, please switch to using a label instead
ERROR: /root/.cache/bazel/_bazel_root/90d880e19ec0b9484a4fbe4609e02db7/external/libcoral/coral/pipeline/BUILD:21:11: @@libcoral//coral/pipeline:common depends on @@glog//:glog in repository @@glog which failed to fetch. no such package '@@glog//': /root/repositories/coral/pycoral_faranick/external/libcoral/third_party/glog/BUILD is not a regular file; if you're using a relative or absolute path for `build_file` in your `new_local_repository` rule, please switch to using a label instead
ERROR: Analysis of target '//src:_pywrap_coral' failed; build aborted: Analysis failed
INFO: Elapsed time: 7.430s, Critical Path: 0.04s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
FAILED: 
    Fetching repository @@bazel_tools~cc_configure_extension~local_config_cc; starting
    Fetching repository @@com_google_absl; starting
make: *** [Makefile:152: pybind] Error 1
root@wvojtas1:~/repositories/coral/pycoral_faranick#
``` 

I never used bazel before, so I am a bit at loss. When I use older bazel, it fails much later (libcoral builds successfully, I can interface with the TPU, but without python bindings).

feranick, 2024-08-07:
Try cleaning up your bazel cache folder (usually in ~/.cache/). Also I'd reclone the repository. Bazel 6.5 is needed, you can find it here: 

https://github.com/bazelbuild/bazel/releases/tag/6.5.0

Again, I can't really answer to any particular compilation issues for Ubuntu 24.04 or Python3.12, because i have no systems that runs on that.

virnik0, 2024-08-07:
`bazel clean --expunge` does exactly that.
```
# bazel --version
bazel 7.2.1
```

It looks like it doesn't even fetch the repositories it is supposed to work with. What approach you take to build natively within 'pycoral' repository context? Maybe I am doing it wrong, last time I was building pycoral, it was nearly two years ago.

virnik0, 2024-08-07:
`# apt-get install bazel=6.5.0` - the build is running now. Will report back once it either fails or succeeds. Thanks for the hint of clang, that is something I didn't noticed. Also, having bazel 6.5.0 seems to be working now.  Looks like I was using either too new, or too old version before (along with gcc instead of clang).

Thanks for those!

feranick, 2024-08-07:
First, bazel 6.5.0 is required for TF 2.17.0. I'd stick to that. 

If you want to do a native compilation (i.e. without docker), I simply run:

```
make pybind
make wheel
```
This with ubuntu:22.04. Works for ma.

feranick, 2024-08-07:
I'd like to remind folks that want to compile these libraries to please read the READMEs. Versions and requirements are listed there. 

https://github.com/feranick/TFlite-builds/blob/main/README.md
https://github.com/feranick/libedgetpu/blob/master/README.md
https://github.com/feranick/pycoral/blob/master/README.md

virnik0, 2024-08-07:
No kidding. Childish mistake on my part, true. First of all, I was using already outdated approach. Second, I was using GCC instead of CLang. And last, I was using either too old (4.5.1) or too new (7.2.1) bazel

I've succeeded in building pycoral. I have uploaded its wheel to my repo for anyone dealing with the same thing to use. 
Link bellow:
https://github.com/virnik0/pycoral_builds

feranick, 2024-08-07:
Cool, glad it worked out.

feranick, 2024-08-07:
Thank you. Yes, that is needed, and will add to the README.md.

[24 more replies]