google-coral/pycoral #140: Unable to install python3-pycoral even install 3.9.19 with pyenv

sangyo1, 2024-03-21:
### Description

I am Ubuntu 20.04 user.
I was trying to install pycoral in raspberry pi.
I install pyenv and installed the python 3.9.19 version. However I set python3.9.19 as global but still unable to install pycoral.
How can I install it? my orginial python version 3.10.X.

[image]

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

USB Accelerator

rahairirani, 2024-03-22:
Try create a virtual env with python3.9.X. I faced the similar problem in raspberry pi 4 64bit OS.

isaacsenn, 2024-04-02:
Hi, were you able to get this sorted? I am facing the same issue.

rahairirani, 2024-04-03:
1. Create virtualenv with python3.9.x
2. Download pycoral-2.0.0-cp39-cp39-linux_aarch64.whl and tflite_runtime-2.5.0.post1-cp39-cp39-linux_x86_64.whl from https://github.com/google-coral/pycoral/releases/tag/v2.0.0
3. pip install pycoral-2.0.0-cp39-cp39-linux_aarch64.whl
4. pip install tflite_runtime-2.5.0.post1-cp39-cp39-linux_x86_64.whl

Not sure manual  whether manual wheel install or virtualenv solved my problem.

isaacsenn, 2024-04-03:
Thanks @rahairirani . Is this only for a 64 bit OS? I'm currently working with 32bit bookworm.

rahairirani, 2024-04-03:
I'm using raspberry pi OS 64bit but the steps also can be used in any 32bit OS.

pagakarthik, 2025-05-24:
There are two parts to this issue. 
1. the python version 
2. the library dependency (specifically numpy)

numpy by default has transitioned to numpy-2.0, so we need to force install an older version of numpy (1.x)  which is required by pycoral

setup that helped me:
- create a virtual environment (with python 3.9 and numpy 1.x)
`conda create --name coral-env python=3.9 "numpy<2" `

- download the latest version of pycoral for your architecture/ OS (e.g. linux_x86_64)
https://github.com/google-coral/pycoral/releases  [e.g. pycoral-2.0.0-cp36-cp36m-linux_x86_64.whl]
run this command
`wget https://github.com/google-coral/pycoral/releases/download/v2.0.0/pycoral-2.0.0-cp36-cp36m-linux_x86_64.whl`
- install the downloaded file
`pip install pycoral-2.0.0-cp36-cp36m-linux_x86_64.whl`

Following that check your device/connections
`conda activate coral-env`
'python3'
`from pycoral.utils.edgetpu import list_edge_tpus`
`print(list_edge_tpus())`