google-coral/pycoral #141: Run YoloV9

atbdeveloper, 2024-03-31:
### Description

Anyone knows about the process for run Yolo V9 inside google coral board?

### Issue Type

Support

### Operating System

Mendel Linux

### Coral Device

Dev Board

zoldaten, 2024-06-03:
almost the same https://docs.ultralytics.com/guides/coral-edge-tpu-on-raspberry-pi/#export-your-model-to-a-edge-tpu-compatible-model

winkmichael, 2024-06-24:
Edge TPU on Raspberry Pi with TensorFlow Lite (New)⭐
The [existing guide](https://coral.ai/docs/accelerator/get-started/) by Coral on how to use the Edge TPU with a Raspberry Pi is outdated, and the current Coral Edge TPU runtime builds do not work with the current TensorFlow Lite runtime versions anymore. In addition to that, Google seems to have completely abandoned the Coral project, and there have not been any updates between 2021 and 2024. This guide will show you how to get the Edge TPU working with the latest versions of the TensorFlow Lite runtime and an updated Coral Edge TPU runtime on a Raspberry Pi single board computer (SBC).