google-coral/pycoral #149: Python 3.12 support

virnik0, 2024-08-06:
### Description

I am trying to build pycoral for use on Ubuntu 24.04 with Python3.12, x86_64 arch.

I have tried few diferent forks, but the latest one and most promissing was faranick's.

Downgrading Python to version 3.11 is not an option given how hardwired 3.12 in the system is. I tried - and only result was corrupted python install and messed-up dependencies (easy to solve with zfs snapshot revert). 

I do have tflite_runtime-2.17.0-cp312-cp312-linux_x86_64.whl wheel build from faranick, and libedgetpu1-max_16.0tf2.17.0-1.ubuntu22.04_amd64.deb (which seems to be working fine on Ubuntu 24.04) from the same author. But I am missing pycoral. While I was able to rebuild libcoral and verify that my coraltpu works, I am unable to use python bindings in anything relevant to it due to absence of pycoral.

I have dowgraded bazel to version 5.4.1, otherwise the build would fail much sooner.

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Relevant Log Output

```shell
~/repositories/coral/pycoral# bazel clean --expunge && TF_PYTHON_VERSION=3.12 CFLAGS='-stdlib=libc++ -std=c++17' make
INFO: Starting clean (this may take a while). Consider using --async if the clean takes more than several minutes.
Starting local Bazel server and connecting to it...
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 1 packages loaded
Loading: 1 packages loaded
PYTHON_BIN_PATH=/usr/bin/python3 bazel build --compilation_mode=opt --copt=-DNPY_NO_DEPRECATED_API=NPY_1_7_API_VERSION --cpu=k8 --linkopt=-L/root/repositories/coral/pycoral/libedgetpu_bin/direct/k8 --linkopt=-l:libedgetpu.so.1 --linkopt=-Wl,--strip-all \
    --embed_label='TENSORFLOW_COMMIT=ad6d8cc177d0c868982e39e0823d0efbfb95f04c' \
    --stamp \
    //src:_pywrap_coral
ERROR: Traceback (most recent call last):
        File ".cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/org_tensorflow/tensorflow/python/tools/api/generator2/generate_api.bzl", line 18, column 34, in <toplevel>
                ApiInfo, _new_api_info = provider(
Error in provider: provider() got unexpected keyword argument 'init'
INFO: Repository rules_proto instantiated at:
  repositories/coral/pycoral/WORKSPACE:129:14: in <toplevel>
  .cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/org_tensorflow/tensorflow/workspace0.bzl:114:17: in workspace
Repository rule http_archive defined at:
  .cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/bazel_tools/tools/build_defs/repo/http.bzl:355:31: in <toplevel>
INFO: Repository rules_cc instantiated at:
  repositories/coral/pycoral/WORKSPACE:126:14: in <toplevel>
  .cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/org_tensorflow/tensorflow/workspace1.bzl:34:14: in workspace
  .cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/com_github_grpc_grpc/bazel/grpc_deps.bzl:158:21: in grpc_deps
Repository rule http_archive defined at:
  .cache/bazel/_bazel_root/be2d1fe448dd770f03a597a2a9fa3688/external/bazel_tools/tools/build_defs/repo/http.bzl:355:31: in <toplevel>
ERROR: Analysis of target '//src:_pywrap_coral' failed; build aborted: error loading package '@org_tensorflow//tensorflow': initialization of module 'tensorflow/python/tools/api/generator2/generate_api.bzl' failed
INFO: Elapsed time: 1.298s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (5 packages loaded, 8 targets configured)
    currently loading: @org_tensorflow//tensorflow
    Fetching https://storage.googleapis.com/mirror.tensorflow.org/github.com/bazelbuild/rules_proto/archive/11bf7c25e666dd7ddacbcd4d4c4a9de7a25175f8.tar.gz
    Fetching https://github.com/bazelbuild/rules_cc/archive/624b5d59dfb45672d4239422fa1e3de1822ee110.tar.gz
make: *** [Makefile:152: pybind] Error 1
```

Ihon75, 2024-09-24:
Same challenge here. 
Compiling for Ubuntu 24.04 is not working

virnik0, 2024-09-24:
> Same challenge here. 
> Compiling for Ubuntu 24.04 is not working

It does. The dev provided link to his git repo. I did as well.
https://github.com/virnik0/pycoral_builds.git

jogugil, 2025-03-28:
Hi there,

I hope you're doing well! First of all, I apologize for the length of this message, but I've been struggling with this for a week now and I don't know what to do anymore.

I was given a "Coral Dev Board" (Raspberry Pi + Edge TPU), but nothing seems to be working for me. Here's what I've tried so far:

a) Initially, I tried connecting the board via OTG mode to my computer, but nothing happens, and I can't figure out how to run a Python script to read sensors, use the camera, or even work with the Edge TPU.

b) Then, I installed Mendel OS on an SD card and tried connecting the board in Host mode, but again, no luck. My computer still doesn't recognize the board, and I'm confused as to why it won't work in Host mode.

c) I decided to try using just the Edge TPU and connect it to my Jupyter notebook on my computer. Unfortunately, I couldn't get it to connect either because I can't install the pycoral library due to multiple dependency issues.

d) As a next step, I installed WSL and I'm trying to create a Docker container with an environment that will at least allow me to run Python scripts for AI models and do inference with the TPU. However, I'm still struggling to create a working Docker container that can load the necessary runtime environment and dependencies for Edge TPU and pycoral.

I was inspired by a post I saw on your website and tried following the instructions to create a Dockerfile, but I still can't get it to work. Specifically, after building the container, when I enter it and try to download the examples, I only get part of them. I'm also unsure whether, after creating the container, I will be able to use it to interact with the Coral board or the Edge TPU.

Here’s my current Dockerfile:

Dockerfile
 
 
```
# Using the official Ubuntu 24.04 base image
FROM ubuntu:24.04

# Setting environment variables to avoid debconf interaction issues
ENV DEBIAN_FRONTEND=noninteractive

# Updating system and installing necessary dependencies
RUN apt-get update && \
    apt-get upgrade -y && \
    apt-get install -y \
    wget \
    python3.12 \
    python3-venv \
    python3-pip \
    curl \
    git \
    build-essential \
    ca-certificates \
    apt-utils \
    clang-18 \
    lsb-release \
    gnupg \
    software-properties-common \
    && rm -rf /var/lib/apt/lists/*

# Installing Bazel 6.5.0
RUN apt-get update && \
    apt-get install -y apt-transport-https curl gnupg && \
    curl -fsSL https://bazel.build/bazel-release.pub.gpg | gpg --dearmor >bazel-archive-keyring.gpg && \
    mv bazel-archive-keyring.gpg /usr/share/keyrings && \
    apt update && \
    apt install -y bazel-6.5.0 && \
    apt update && \
    apt full-upgrade -y && \
    apt clean

# Installing tflite_runtime and pycoral from provided .whl files
RUN pip install --upgrade pip setuptools && \
    pip install \
    https://github.com/virnik0/pycoral_builds/raw/main/tflite_runtime-2.18.0-cp312-cp312-linux_x86_64.whl && \
    pip install \
    https://github.com/virnik0/pycoral_builds/raw/main/pycoral-2.0.2-cp312-cp312-linux_x86_64.whl

# Setting the working directory
WORKDIR /app

# Installing Python dependencies
RUN python3 -m venv /env && \
    . /env/bin/activate && \
    pip install --upgrade pip setuptools

# Copying application code to the container
COPY . /app

# Installing Python dependencies from requirements.txt
RUN . /env/bin/activate && \
    pip install -r requirements.txt

# Exposing the port if necessary
EXPOSE 8000

# Default command
CMD ["/bin/bash"]
```